kmolan/multicalc-rust #1: accept closure rather than a function

eweca-d, 2024-06-17:
Thank you for doing this great job!

I am trying to make use of this crate to generate jacobian matrix  for curve fitting of arbitrary functions with crate `levenberg-marquardt`. Therefore, I need to derive jacobian matrix based on the generated closures that capture input xdata in runtime. Do you have any plan to support closure as the element of `function_matrix` in the module of `Jacabian`?

kmolan, 2024-06-17:
Hi, can you provide a self-contained example so I can reproduce the error? It was my understanding that `jacobian` should not have any problem accepting a vector of closures.

eweca-d, 2024-06-17:
My bad. Just find I did not remember to annotate like `Vec<&dyn Fn(&[f64; M]) -> f64>`. Thank you for your great job again!

kmolan, 2024-06-17:
No problem. Glad it all worked out. Closing the issue as resolved.

closed 2024-06-17 by kmolan