kmolan/multicalc-rust #2: get jacobian and hessian not only row by row

eweca-d, 2024-06-18:
I am using this crate to do curve fitting with 'nalgebra' that is an algebra crate. For all I know, most algebra library tends to use column-major matrix and store matrix in a vec storage. Hence, I think it is better to provide new api or pass the paramter to control the output matrix shape to be:

- column-major: `[[T; nrows]; ncols]` 
- contiguous column by column: `[T; nrows * ncols]`
- contiguous row by row: `[T; nrows * ncols]`

Thank you for your great job. It works very well in my crate right now.

kmolan, 2024-06-18:
Hi @eweca-d I see only three possible output matrices. What's the difference between contiguous column by column and contiguous row by row?

eweca-d, 2024-06-18:
"contiguous column by column" is the flatten array of column-major matrix, "contiguous row by row" is the flatten array of row-major matrix. There can be constructed by:

```rust
pub fn get_custom<T: ComplexFloat, const NUM_FUNCS: usize, const NUM_VARS: usize>(function_matrix: &[&dyn Fn(&[T; NUM_VARS]) -> T; NUM_FUNCS], vector_of_points: &[T; NUM_VARS], step_size: f64, mode: mode::DiffMode) -> Result<[[T; NUM_VARS]; NUM_FUNCS], ErrorCode>
{
    if function_matrix.is_empty()
    {
        return Err(ErrorCode::VectorOfFunctionsCannotBeEmpty);
    }

    // row-major
    let mut result = [[T::zero(); NUM_VARS]; NUM_FUNCS];
    for row_index in 0..NUM_FUNCS
    {
        for col_index in 0..NUM_VARS
        {
            result[row_index][col_index] = single_derivative::get_partial_custom(&function_matrix[row_index], col_index, vector_of_points, step_size, mode)?;
        }
    }

    // col-major
    let mut result = [[T::zero(); NUM_FUNCS]; NUM_VARS];
    for col_index in 0..NUM_VARS
    {
        for row_index in 0..NUM_FUNCS
        {
            result[col_index][row_index] = single_derivative::get_partial_custom(&function_matrix[row_index], col_index, vector_of_points, step_size, mode)?;
        }
    }

    // contiguous row by row
    let mut result = [T::zero(); NUM_VARS * NUM_FUNCS];
    for row_index in 0..NUM_FUNCS
    {
        for col_index in 0..NUM_VARS
        {
            result[row_index * NUM_VARS + col_index] = single_derivative::get_partial_custom(&function_matrix[row_index], col_index, vector_of_points, step_size, mode)?;
        }
    }

    // contiguous col by col
    let mut result = [T::zero(); NUM_FUNCS * NUM_VARS];
    for col_index in 0..NUM_VARS
    {
        for row_index in 0..NUM_FUNCS
        {
            result[col_index * NUM_FUNCS + row_index] = single_derivative::get_partial_custom(&function_matrix[row_index], col_index, vector_of_points, step_size, mode)?;
        }
    }
    
    return Ok(result);
}
```

----------------------------------------------------------

As for the heap allocations, I do not think it is a problem. The current implement can just work for a small size Jacobian matrix. However, if I try to fit a curve with 20000 data points, the stack will overflow. Furthermore, the vector has no significant impact on the performance if the size is relatively large. I recommend replace the array of array by `nalgebra` Matrix. It provides static allocated and dynamic matrix for the use of both small and large matrix.

eweca-d, 2024-06-18:
If we want a custom funtion to allocate the data in stack, I guess we should do some unsafe thing like returning a pointer `*mut T` of `[T; N * M]`. Then, we can convert `*mut T` to the shape we desired. However, it can only work if the data stored in stack would not be deconstructed after returing the stack pointer.

kmolan, 2024-06-18:
You point about stack overflow is a good one. Unfortunately even with nalgebra the static matrices (SMatrix) would soon overflow with 20000 points. What we really need here is a DMatrix for heap allocation. I will soon push a change for jacobians and hessians to use SMatrix. I'm choosing to keep the same output matrix structure for API compatibility. It should be trivial for matrix shape changes using the nalgebra directly.

I have also started a bigger task of providing methods using vectors and DMatrices as an optional feature of the crate. That one might take some time.

eweca-d, 2024-06-18:
Thank you for your effort! I really expect to enable the new features in my work.

kmolan, 2024-06-18:
https://github.com/kmolan/multicalc-rust?tab=readme-ov-file#16-experimental
Chose to keep the matrix shape the same for API compatibility. Tried nalgebra DMatrix but num-complex traits don't play nice with it (they don't implement the Debug trait). 

Long term solution is to hand-roll my own numeric trait, but for now this should atleast help you with stack overflows.

eweca-d, 2024-06-18:
Thank you so much! It will really help if I try to derive a big Jacobian matrix in my work!